simplefoc/Arduino-FOC #462: [BUG] MCPWM error on ESP32-S3

sanjay-swain, 2025-04-03:
Hello, I just bought a simplefocmini on for the first time and uploaded the open loop motor control example but was getting the following error:
```
ESP32-DRV: ERROR - Could not initialize the timer in group: 0
E (5) mcpwm: mcpwm_new_timer(106): register back up is not supported
Driver Init Failed
```

I am using Arduino IDE. Any idea what is causing this?

runger1101001, 2025-04-10:
Thanks for reporting it. We will fix it for the next release…

ogiel, 2025-06-06:
This is actually wildly frustrating and PLEASE someone help me out here. I think I have now wasted more than a full day going back and forth between different errors that now finally appear to be related to this issue.

I try to setup PlatformIO / ESP32 / AS5600 / BLDC (GBM2804) combination and I'm new to ESP32 and PlatformIO.

I've tried endless combinations in my platform.ini and all combinations give some error but the error is dependent on the combination.

Everywhere people mention ESP-IDF (5.4, 5.3 idk) or ESP core versions (3.2.0??) but nowhere people specify these versions in their platform.ini becuase (I believe) they are inside the platform of choice. 

Some combos:
This:
```
platform = espressif32@6.4.0
lib_deps = askuric/Simple FOC@^2.3.4
```

Gives:
`.pio/libdeps/esp32dev/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcpwm_mcu.cpp:9:2: error: #error SimpleFOC: ESP-IDF version 4 or lower detected. Please update to ESP-IDF 5.x and Arduino-esp32 3.0 (or higher)`

This:
```
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
lib_deps = askuric/Simple FOC@^2.3.4
```

Gives:
`E (14) mcpwm: mcpwm_new_timer(106): register back up is not supported`

I think I'm close now but this is such a waste of time. What is a working combination and HOW can someone new to SimpleFOC / ESP32 find out without having to go insane because using the latest stable version for platform and simplefoc is not compatible?

teebrin, 2025-06-19:
I went back to an earlier version than the stable one provided by pioarduino but still more recent than the official espressif32 platform.
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip`
Seems to work for me with Simple FOC 2.3.4. Hope that helps!

Orusdsm, 2025-08-04:
Hola he comprado una tarjeta controladora china parecida a la fagor 

[image] le descargue la libreria pero tengo priblema con la identificación de los pines,  revise si hay conmutación en las salidas e idemtifique qué con los pines 35 37 y 28  con estos pines hay conmutación en las 3 salidad pero el motor solo se energiza intermitente mente en una sola posición y no gira, ya pedí la documentación a proveedor pero no hay respuesta alguien ya a trabajado con esta tarjeta qué me pueda ayudar ?

closed 2025-08-06 by runger1101001